alex3dbros/megacnc #1: [MEGA CNC Alpha] Missing temperature graph

kamilzierke, 2024-02-21:
Missing temperature graph

[image]

kamilzierke, 2024-02-21:
[image]
Different example of Apexcharts config, improving visibility of Voltage/Current. Temperature in the brush graph below main one.

[image]

Just an example.

kamilzierke, 2024-02-21:
The above was actually an Apexcharts.js implementation in Home Assistant done by https://github.com/RomRider/apexcharts-card
The card translated to Apexcharts.js should look something like this:

var options = {
  chart: {
    type: 'line',
    height: 350,
    toolbar: {
      show: true,
      tools: {
        selection: true
      }
    },
    brush: {
      enabled: true,
      target: 'chart',
    },
    selection: {
      enabled: true,
      xaxis: {
        min: new Date('2023-01-01').getTime(),
        max: new Date('2023-01-01').getTime() + 3600000 // 1 hour in milliseconds
      }
    }
  },
  series: [
    {
      name: 'Voltage',
      data: [] // Populate with actual data for 'sensor.mcc_cell_01_vlt'
    },
    {
      name: 'Temperature',
      data: [] // Populate with actual data for 'sensor.mcc_cell_01_tmp'
    },
    {
      name: 'Current',
      data: [] // Populate with actual data for 'sensor.mcc_cell_01_amp'
    }
  ],
  title: {
    text: 'Cell 01',
    align: 'left'
  },
  stroke: {
    width: 2
  },
  markers: {
    size: 0
  },
  tooltip: {
    shared: true,
    intersect: false,
    x: {
      format: 'dd MMM - HH:mm'
    }
  },
  xaxis: {
    type: 'datetime'
  },
  yaxis: [{
    title: {
      text: 'Voltage',
    },
  }, {
    opposite: true,
    title: {
      text: 'Temperature'
    }
  }, {
    opposite: true,
    title: {
      text: 'Current'
    }
  }],
  legend: {
    show: true
  }
};

var chart = new ApexCharts(document.querySelector("#chart"), options);
chart.render();

alex3dbros, 2024-02-23:
I added the temperature to replace the capacity. I will close the issue for now as I have an open issue with graph becoming unresponsive.

closed 2024-02-23 by alex3dbros