alex3dbros/megacnc #4: [MEGA CNC Alpha] All measurements resets before taking out the measured cell from the slot.

kamilzierke, 2024-02-21:
All measurements resets before taking out the measured cell from the slot.
The workflow was started, ESR measurement, capacity and bringing the cell to the storage voltage were completed. 
Before the cell was taken out from the slot everything was reset:

[image]
[image]

alex3dbros, 2024-02-23:
The reset could happen only if the device actually reset. Please try to reproduce or find a way to check this

kamilzierke, 2024-02-23:
[image]

Unfortunately I don't know how to replicate it, it doesn't happen all the time, rather random.
The device was not reset (there was ongoing charging on slot 8)

kamilzierke, 2024-02-23:
Another observation: the mCAP timer was not reset and continue normally.

[image]

Adding also CSV with history data gathered from API.
[history.csv](https://github.com/alex3dbros/megacnc/files/14390870/history.csv)

kamilzierke, 2024-02-23:
Probably we should change the classification of this error from MEGA CNC to FW issue; I was running mCAP through and API and gathering data the same way, MEGA CNC was not involved directly.

alex3dbros, 2024-02-24:
I think a charge command was sent to your device. That will reset the capacity but not the counter. That explains the reason for showing sudden voltage raise. You can reproduce this behaviour by sending a charge command after store state is reached. At that point the capacity counter is being reset.

kamilzierke, 2024-02-24:
I wasn't able to replicate the same situation when operating only through the MCC device menu, but I was able to replicate the issue with resetting capacity measurement:

[image]

After 2hrs of sitting in the slot (1:59h for cell 1, 2:06hr for cell 8) the counter was reset and the cell status changed to "Regular cell"

alex3dbros, 2024-02-24:
Try to make a video of the device display to make sure it does not reset. Also, are you doing any api request with your custom monitoring software directly to megacell charger? There's a certain number of cells you can request info for from the device to avoid memory overflow.

kamilzierke, 2024-02-24:
In the previous example I was running the below:

rest:
  - resource: http://192.168.1.156/api/get_cells_info
    method: POST
    headers:
      Content-Type: "application/json"
    payload: '{"start": 1, "end": 16}'
    scan_interval: 15
    timeout: 30

So basically I'm requesting every cell info every 15s.

alex3dbros, 2024-02-24:
Ok, that's the problem. Split that request in 2 different requests, 1 to 8 and 9 to 16 and you won't have that issue.

closed 2024-02-24 by alex3dbros