alex3dbros/megacnc #5: bug in device details

imkermichel, 2024-02-24:
when running macro with 1 cell the interface correctly tells 15 cells not inserted but all other types stay at 0.
[image]

alex3dbros, 2024-02-24:
Hi, can you explain the issue ? I don't understand what the problem is. You have 16 cells inserted or just 1 cell inserted? If it's just one cell inserted, then it make sense not to update the rest when cell 1 starts mcap test

imkermichel, 2024-02-24:
I have 1 cell inserted and thought it would also show 1 regular cell

alex3dbros, 2024-02-24:
The device was stopped at that time ? It finished the mcap or was running ? It takes a few seconds to get status update from the device. Make sure you have a good wifi connectivity. Try resetting the device and see if you can reproduce this issue and write me how. Also I need to know your hardware type and firmware version running on the megacell charger. I display in the software what the megacell charger says.

imkermichel, 2024-02-24:
[image]
[image]
picture 1: insert new cells (2)
picture 2: started macro with 2 cells
in the summary it only tells the 14 not inserted and not 2 charging.

Kripits, 2024-02-25:
Hi, I have similar issue, I'm doing pack series capacity measurement via macros, I have a setup with 6 slots combined. It shows first 6 slots like one slot and when it goes to almost full charge, the first slot on hardware stops charging and other slots keep low current charge to not get overcurrent (which is normal), but the APP shows CHARGED state although it is still charging with some current which you can see at CURRENT column.
[image]
It would be nice to show correct state guided by all the slots which are connected in group an to show (in my case) only two groups plus the remaining slots 6+6 (groups) +1+1+1+1 (separate slots)