alex3dbros/megacnc #6: MegaCellCharger firmware issue

JamesJamJar, 2024-04-20:
So after updating firmware for MCC regular eairly batch 
it doesn't boot, causes green LEDs to randomly blink an some stay on. I endup having to revert back to the last versions firmware 4.3
Is there a fix? Or am i doing something wrong?

JamesJamJar, 2024-04-22:
I am now no longer able to charge getting a FAIL! FAIL! on screen from the firmware downloaded from your megacell website. 

Any assistance would be great. Thanks

alex3dbros, 2024-04-22:
Hi, I am not sure which megacell website you are referring to. The fail fail is a message that was included in the old megacell charger firmware. Download the new firmware from the firmwares folder:
https://github.com/alex3dbros/megacnc/tree/main/Megacell%20Firmwares

JamesJamJar, 2024-04-22:
https://megacellmonitor.com/

JamesJamJar, 2024-04-22:
Hi Alex, Thanks for the response. 

So the issue i have with https://github.com/alex3dbros/megacnc/tree/main/Megacell%20Firmwares is stated in the first message. 
it doesn't boot, causes green LEDs to randomly blink an some stay on and doesn't do anything else.

closed 2024-04-22 by alex3dbros

JamesJamJar, 2024-04-22:
So the device was working absolutely fine before. But i can no longer find the lastest firmware from the old MCM setup. Is there somewhere i can download it from or you could point to me. As whats on here https://megacellmonitor.com/ is the very old 4.3 version. This does not work causing the FAIL! error. I would like to just go back as CellDoctor.production.ino.1.0.2_MCCRegular.bin clearly does not work on my version of the MCC. 

Thanks again.

JamesJamJar, 2024-04-24:
Any update on the last comment Alex?

JamesJamJar, 2024-04-25:
@alex3dbros